unmannedlab/RELLIS-3D #3: Ontology mismatch

etevere11, 2021-01-20:
Hi,

I am using the HRNet model to test on a different dataset and came across a couple of issues when referencing the ontology documentation:

1. In ontology.csv you list a class 32 [deep water] which is not included in the ontology.yaml - is this label included in the dataset?
2. When I use the HRNet test function and visualize results, the model output/prediction contains labels that do not exist for the dataset like 2 and 13. Any idea why this might be happening or how to fix this issue?

maskjp, 2021-01-20:
Hi, @etevere11 ,

Thank you for your interest in RELLI-3D. Sorry for the mismatch, we don't have a deep water class in the annotation. Please ignore the class. We just updated the csv file, thank you for your help.

If you used the test script of HRNet, the output labels are mapped between 0 to the number of classes. You can use this [function](https://github.com/unmannedlab/RELLIS-3D/blob/794089082b7803443266a65336e959a65ee0fd98/benchmarks/HRNet-Semantic-Segmentation-HRNet-OCR/lib/datasets/rellis.py#L104) to remap it. 

Hope this answer solves your problem.

etevere11, 2021-01-26:
Hi @maskjp,

Thanks for the updated information. In regards to the label_mapping function that you linked, is there a reason that both class 0 [void] and class 1 [dirt] are remapped to the same value of 0? Shouldn't these both continue to be included in the image, or does your model remove the void class from consideration, and if so, are areas in the original image that are labeled as being void interpreted as dirt then?
[image]

maskjp, 2021-01-26:
Hi, @etevere11,
We mapped the dirt to void because the number of the dirt pixels is too small to learn. When training and evaluating, we ignore the void class, so the dirt is also ignored.