unmannedlab/RELLIS-3D #4: HRnet with lr1e-3

xyaoab, 2021-01-27:
May I know how to reproduce the results of HRNet-OCR benchmark? With the model you uploaded(lr=1e-2), on the validation set, class iou is pretty low ([0. 0.39814434 0.42628074 0. 0. 0.94645624 0.  0. 0.  0. 0. 0. 0. 0.09216522 0.  0. 0. 0.45838588 0.]). Any suggestions?

maskjp, 2021-01-27:
Hi, @xyaoab,

Thank you for your interest in RELLIS-3D! If you used the provided script, then the saved prediction id is from 0 to num_classes. When you compute the iou, you may need to use this [function](https://github.com/unmannedlab/RELLIS-3D/blob/794089082b7803443266a65336e959a65ee0fd98/benchmarks/HRNet-Semantic-Segmentation-HRNet-OCR/lib/datasets/rellis.py#L104) to convert the label into the original id based on this [mapping](https://github.com/unmannedlab/RELLIS-3D/blob/794089082b7803443266a65336e959a65ee0fd98/benchmarks/HRNet-Semantic-Segmentation-HRNet-OCR/lib/datasets/rellis.py#L51).

Hope this helps!

Best wishes!

Peng Jiang

jdonovanCS, 2021-06-17:
I am also having this trouble, but I will try the conversion you mentioned above and report back. Additionally, I was trying to run the Evaluate_img python notebook, but the folder structure of the dataset after I downloaded it does not match up. Maybe I downloaded the wrong zip file. I also didn't know if there was a way to output the predictions to load into that python notebook. I'm going to continue looking. My thought is maybe I just skipped some steps in my haste.

maskjp, 2021-06-18:
> I am also having this trouble, but I will try the conversion you mentioned above and report back. Additionally, I was trying to run the Evaluate_img python notebook, but the folder structure of the [...]

Hi, @jdonovanCS,

This is an old issue. We changed the code and it can evaluate the label without conversion. 

On the other hand, what folder structure did you get? Can you tell me more detail?

Best wishes!

jdonovanCS, 2021-06-18:
I think I actually figured out some of what was going on. After I downloaded the Full Images and the Full Images ID Format separately. I just needed to copy the Rellis-3D folder from one into the other and that made it correct. The issue I'm facing now is two-fold. 1) I keep running out of CUDA Memory even though I have a NVIDIA Mobile 1070 which has at least as much GPU RAM as the 1050 TI that was used by the project (I believe) and 2) I can't figure out how to produce actual prediction images to point the Evaluate_img python notebook at.

maskjp, 2021-06-18:
> I think I actually figured out some of what was going on. After I downloaded the Full Images and the Full Images ID Format separately. I just needed to copy the Rellis-3D folder from one into the [...]

Hi, @jdonovanCS,

Try to reduce the batch size and input image size in the configuration file. As I show in the previous answer, the input size was reduced to 512x512 in order to perform the inference on a 1050 Ti.

The models were trained on a more powerful workstation with two GPUs. We tested inference on 1050Ti because we want to see whether we can use the model on a real robot with a small GPU.

jdonovanCS, 2021-06-18:
Oh ok. Cool. I misunderstood. Thank you!