unmannedlab/RELLIS-3D #6: Problems to generate the benchmark results

SvenMuc, 2021-04-21:
Hi,
I'm currently investigating the RELLIS-3D dataset and trying to generate the benchmark results with the SalsaNext model.

First I generated all predictions (`.label` files) by the `run_salsanext_eval.sh` script.
Afterwards I used the `eval.sh` and the `evaluate_iou.py` script directly  in the SalsaNext benchmark directory to generate the IoU values. I just receive 0 IoU for all classes.

How do I generate the IoU values, correctly? 

```
Evaluating sequences: 
['train', 'valid', 'test'] set:
Acc avg 0.000
IoU avg 0.000
IoU class 1 [car] = 0.000
IoU class 2 [bicycle] = 0.000
IoU class 3 [motorcycle] = 0.000
IoU class 4 [truck] = 0.000
IoU class 5 [other-vehicle] = 0.000
IoU class 6 [person] = 0.000
IoU class 7 [bicyclist] = 0.000
IoU class 8 [motorcyclist] = 0.000
IoU class 9 [road] = 0.000
IoU class 10 [parking] = 0.000
IoU class 11 [sidewalk] = 0.000
IoU class 12 [other-ground] = 0.000
IoU class 13 [building] = 0.000
IoU class 14 [fence] = 0.000
IoU class 15 [vegetation] = 0.000
IoU class 16 [trunk] = 0.000
IoU class 17 [terrain] = 0.000
IoU class 18 [pole] = 0.000
IoU class 19 [traffic-sign] = 0.000
```
Thanks for you support.

maskjp, 2021-04-22:
Hi, @SvenMuc Thank you for your interest in RELLIS-3D. The ```eval.sh``` is not working for evaluation.  I just added new [scripts](https://github.com/unmannedlab/RELLIS-3D/blob/main/utils/Evaluate_pt.ipynb) for evaluation please use them. Thanks!

SvenMuc, 2021-04-23:
@maskjp Thank you for your quick reaction. Works perfectly.

closed 2021-04-23 by SvenMuc

Varatharajan-Raja, 2022-04-20:
I'm also trying to evaluate RELLIS-3D dataset with SalsaNext. @maskjp 
I'm new to this field. I have the following questions.

I could see that the run_salsanext_eval.sh had been changed after the new script you have mentioned. So, can I try evaluating with run_salsanext_eval.sh ?

Moreover, I'm not able to run the run_salsanext_eval.sh in my system, Where I only have a intel HD 520 graphics for video. 

Method 1:

I installed torch 1.6.0 for the CPU version and executed the code. 
export CUDA_VISIBLE_DEVICES="0"

[image]

Method 2:

I installed torch 1.6.0 for the CUDA 10.1 version and executed the code
I made the CUDA_VISIBLE_DEVICES="" to take the cpu and executed the code. 

[image]

Also How did you produce 2D image of the point cloud data as you have shown in your predictions result images along with your pretrained salsanext model ?

Thanks

maskjp, 2022-04-21:
Hi, @Varatharajan-Raja ,

Your GPU memory is only 2 GB. You might want to reduce the input size of the image. 

If you want to project the lidar label on the image, you can use this [example](https://github.com/unmannedlab/RELLIS-3D/blob/main/utils/lidar2img.ipynb).

I hope this helps!

Best wishes!