unmannedlab/RELLIS-3D #8: pylon_camera_node plays greyscale image

Tiga002, 2021-05-31:
Thank you for your wonderful effort! 
So I downloaded  example_synced.bag (Synced data (60 seconds example 2 GB): includes synced /os1_cloud_node/points, /pylon_camera_node/camera_info and /pylon_camera_node/image_raw) and played it in my local machine with RVIZ.  

However, I found that Images from (/pylon_camera_node/image_raw) are all greyscaled images (mono) instead of RGB images which is not the same as the dataset(RELLIS-3D/0000x/pylon_camera_node/). 

I tried the rosbags for Sequence 00000 as well. RGB images are not recorded under topic (/pylon_camera_node/image_raw). The only RGB images I found is from the stereo camera (/nerian/left/image_raw or /nerian/right/image_raw), which the resolution is not the same as the training dataset.

maskjp, 2021-06-01:
Hi, @Tiga002 

Thank you for your interest in RELLIS-3D.

You need to rectify the /pylon_camera_node/image_raw image using [image_proc](http://wiki.ros.org/image_proc) package.
Please try the following commands while you play the rosbag.

`ROS_NAMESPACE=/pylon_camera_node rosrun image_proc image_proc`

I hope this helps!

Best wishes!

Peng Jiang

Tiga002, 2021-06-01:
Oh yeah! It works now! Thank you so much for your fast response and help!
Good Luck with your work!

closed 2021-06-01 by Tiga002