unmannedlab/RELLIS-3D #10: Resizing images from .yaml

jdonovanCS, 2021-06-23:
Hi I was trying to resize the images from the .yaml file, but it is not working the way I thought it would.

I changed TEST.IMAGE_SIZE to 512 x 512 and the TEST.BASE_SIZE to 512.
The images aren't getting resized when processed through the test function in function.py. They are still at 1200 x 1920.

Am I missing a step? Or do I need to change something in the way the dataset is being set up?

```test_dataset = eval('datasets.'+config.DATASET.DATASET)(
                        root=config.DATASET.ROOT,
                        list_path=config.DATASET.TEST_SET,
                        num_samples=None,
                        num_classes=config.DATASET.NUM_CLASSES,
                        multi_scale=False,
                        flip=False,
                        ignore_label=config.TRAIN.IGNORE_LABEL,
                        base_size=config.TEST.BASE_SIZE,
                        crop_size=test_size,
                        downsample_rate=1)```

maskjp, 2021-06-23:
Hi, @jdonovanCS,

I am sorry. I think there's a bug in our code. I just fixed it. Please try to pull the new code 005953fa1936557bda6f139db18acfe91e3f4e19. I tested it on my computer; it outputs resized labels.

Hope it helps!

Best wishes!

jdonovanCS, 2021-06-23:
@maskjp,

No problem. Thanks for the super quick fix. It worked on my end as well.

closed 2021-06-23 by jdonovanCS