unmannedlab/RELLIS-3D #11: How to load trained SalsaNext model for inference

SvenMuc, 2021-07-13:
Hello,
I'm currently trying to load your pretrained model for inference.

```
print("Initialize SalsaNext model...")
number_classes = 15      # read from yaml config file in my code
model = SalsaNext(number_classes)

print("Load model checkpoint...")
checkpoint = torch.load(FLAGS.trained_model)
model.load_state_dict(checkpoint['state_dict'], strict=True)
```

This gives me the following error messages. 

```
Load model checkpoint...
Traceback (most recent call last):
  File "/home/i009658/workspace/AI_EnvPerception/datasets/RELLIS-3D/benchmarks/SalsaNext/train/tasks/semantic/convert_pytorch_model_to_torch_script_model.py", line 91, in <module>
    model.load_state_dict(checkpoint['state_dict'], strict=True)
  File "/home/i009658/miniconda3/envs/salsanext/lib/python3.7/site-packages/torch/nn/modules/module.py", line 777, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for SalsaNext:
	Missing key(s) in state_dict: "downCntx.conv1.weight", "downCntx.conv1.bias", "downCntx.conv2.weight", "downCntx.conv2.bias", "downCntx.bn1.weight", "downCntx.bn1.bias", "downCntx.bn1.running_mean", "downCntx.bn1.running_var", "downCntx.conv3.weight", "downCntx.conv3.bias", "downCntx.bn2.weight", "downCntx.bn2.bias", "downCntx.bn2.running_mean", "downCntx.bn2.running_var", "downCntx2.conv1.weight", "downCntx2.conv1.bias", "downCntx2.conv2.weight", "downCntx2.conv2.bias", "downCntx2.bn1.weight", "downCntx2.bn1.bias", "downCntx2.bn1.running_mean", "downCntx2.bn1.running_var", "downCntx2.conv3.weight", "downCntx2.conv3.bias", "downCntx2.bn2.weight", "downCntx2.bn2.bias", "downCntx2.bn2.running_mean", "downCntx2.bn2.running_var", "downCntx3.conv1.weight", "downCntx3.conv1.bias", "downCntx3.conv2.weight", "downCntx3.conv2.bias", "downCntx3.bn1.weight", "downCntx3.bn1.bias", "downCntx3.bn1.running_mean", "downCntx3.bn1.running_var", "downCntx3.conv3.weight", "downCntx3.conv3.bias", "downCntx3.bn2.weight", "downCntx3.bn2.bias", "downCntx3.bn2.running_mean", "downCntx3.bn2.running_var", "resBlock1.conv1.weight", "resBlock1.conv1.bias", "resBlock1.conv2.weight", "resBlock1.conv2.bias", "resBlock1.bn1.weight", "resBlock1.bn1.bias", "resBlock1.bn1.running_mean", "resBlock1.bn1.running_var", "resBlock1.conv3.weight", "resBlock1.conv3.bias", "resBlock1.bn2.weight", "resBlock1.bn2.bias", "resBlock1.bn2.running_mean", "resBlock1.bn2.running_var", "resBlock1.conv4.weight", "resBlock1.conv4.bias", "resBlock1.bn3.weight", "resBlock1.bn3.bias", "resBlock1.bn3.running_mean", "resBlock1.bn3.running_var", "resBlock1.conv5.weight", "resBlock1.conv5.bias", "resBlock1.bn4.weight", "resBlock1.bn4.bias", "resBlock1.bn4.running_mean", "resBlock1.bn4.running_var", "resBlock2.conv1.weight", "resBlock2.conv1.bias", "resBlock2.conv2.weight", "resBlock2.conv2.bias", "resBlock2.bn1.weight", "resBlock2.bn1.bias", "resBlock2.bn1.running_mean", "resBlock2.bn1.running_var", "resBlock2.conv3.weight", "resBlock2.conv3.bias", "resBlock2.bn2.weight", "resBlock2.bn2.bias", "resBlock2.bn2.running_mean", "resBlock2.bn2.running_var", "resBlock2.conv4.weight", "resBlock2.conv4.bias", "resBlock2.bn3.weight", "resBlock2.bn3.bias", "resBlock2.bn3.running_mean", "resBlock2.bn3.running_var", "resBlock2.conv5.weight", "resBlock2.conv5.bias", "resBlock2.bn4.weight", "resBlock2.bn4.bias", "resBlock2.bn4.running_mean", "resBlock2.bn4.running_var", "resBlock3.conv1.weight", "resBlock3.conv1.bias", "resBlock3.conv2.weight", "resBlock3.conv2.bias", "resBlock3.bn1.weight", "resBlock3.bn1.bias", "resBlock3.bn1.running_mean", "resBlock3.bn1.running_var", "resBlock3.conv3.weight", "resBlock3.conv3.bias", "resBlock3.bn2.weight", "resBlock3.bn2.bias", "resBlock3.bn2.running_mean", "resBlock3.bn2.running_var", "resBlock3.conv4.weight", "resBlock3.conv4.bias", "resBlock3.bn3.weight", "resBlock3.bn3.bias", "resBlock3.bn3.running_mean", "resBlock3.bn3.running_var", "resBlock3.conv5.weight", "resBlock3.conv5.bias", "resBlock3.bn4.weight", "resBlock3.bn4.bias", "resBlock3.bn4.running_mean", "resBlock3.bn4.running_var", "resBlock4.conv1.weight", "resBlock4.conv1.bias", "resBlock4.conv2.weight", "resBlock4.conv2.bias", "resBlock4.bn1.weight", "resBlock4.bn1.bias", "resBlock4.bn1.running_mean", "resBlock4.bn1.running_var", "resBlock4.conv3.weight", "resBlock4.conv3.bias", "resBlock4.bn2.weight", "resBlock4.bn2.bias", "resBlock4.bn2.running_mean", "resBlock4.bn2.running_var", "resBlock4.conv4.weight", "resBlock4.conv4.bias", "resBlock4.bn3.weight", "resBlock4.bn3.bias", "resBlock4.bn3.running_mean", "resBlock4.bn3.running_var", "resBlock4.conv5.weight", "resBlock4.conv5.bias", "resBlock4.bn4.weight", "resBlock4.bn4.bias", "resBlock4.bn4.running_mean", "resBlock4.bn4.running_var", "resBlock5.conv1.weight", "resBlock5.conv1.bias", "resBlock5.conv2.weight", "resBlock5.conv2.bias", "resBlock5.bn1.weight", "resBlock5.bn1.bias", "resBlock5.bn1.running_mean", "resBlock5.bn1.running_vLoad model checkpoint...
Traceback (most recent call last):
  File "/home/i009658/workspace/AI_EnvPerception/datasets/RELLIS-3D/benchmarks/SalsaNext/train/tasks/semantic/convert_pytorch_model_to_torch_script_model.py", line 91, in <module>
    model.load_state_dict(checkpoint['state_dict'], strict=True)
  File "/home/i009658/miniconda3/envs/salsanext/lib/python3.7/site-packages/torch/nn/modules/module.py", line 777, in load_state_dict
    self.__class__.__name__, "\n\t".join(error_msgs)))
RuntimeError: Error(s) in loading state_dict for SalsaNext:
	Missing key(s) in state_dict: "downCntx.conv1.weight", "downCntx.conv1.bias", "downCntx.conv2.weight", "downCntx.conv2.bias", "downCntx.bn1.weight", "downCntx.bn1.bias", "downCntx.bn1.running_mean", "downCntx.bn1.running_var", "downCntx.conv3.weight", "downCntx.conv3.bias", "downCntx.bn2.weight", "downCntx.bn2.bias", "downCntx.bn2.running_mean", "downCntx.bn2.running_var", "downCntx2.conv1.weight", "downCntx2.conv1.bias", "downCntx2.conv2.weight", "downCntx2.conv2.bias", "downCntx2.bn1.weight", "downCntx2.bn1.bias", "downCntx2.bn1.running_mean", "downCntx2.bn1.running_var", "downCntx2.conv3.weight", "downCntx2.conv3.bias", "downCntx2.bn2.weight", "downCntx2.bn2.bias", "downCntx2.bn2.running_mean", "downCntx2.bn2.running_var", "downCntx3.conv1.weight", "downCntx3.conv1.bias", "downCntx3.conv2.weight", "downCntx3.conv2.bias", "downCntx3.bn1.weight", "downCntx3.bn1.bias", "downCntx3.bn1.running_mean", "downCntx3.bn1.running_var", "downCntx3.conv3.weight", "downCntx3.conv3.bias", "downCntx3.bn2.weight", "downCntx3.bn2.bias", "downCntx3.bn2.running_mean", "downCntx3.bn2.running_var", "resBlock1.conv1.weight", "resBlock1.conv1.bias", "resBlock1.conv2.weight", "resBlock1.conv2.bias", "resBlock1.bn1.weight", "resBlock1.bn1.bias", "resBlock1.bn1.running_mean", "resBlock1.bn1.running_var", "resBlock1.conv3.weight", "resBlock1.conv3.bias", "resBlock1.bn2.weight", "resBlock1.bn2.bias", "resBlock1.bn2.running_mean", "resBlock1.bn2.running_var", "resBlock1.conv4.weight", "resBlock1.conv4.bias", "resBlock1.bn3.weight", "resBlock1.bn3.bias", "resBlock1.bn3.running_mean", "resBlock1.bn3.running_var", "resBlock1.conv5.weight", "resBlock1.conv5.bias", "resBlock1.bn4.weight", "resBlock1.bn4.bias", "resBlock1.bn4.running_mean", "resBlock1.bn4.running_var", "resBlock2.conv1.weight", "resBlock2.conv1.bias", "resBlock2.conv2.weight", "resBlock2.conv2.bias", "resBlock2.bn1.weight", "resBlock2.bn1.bias", "resBlock2.bn1.running_mean", "resBlock2.bn1.running_var", "resBlock2.conv3.weight", "resBlock2.conv3.bias", "resBlock2.bn2.weight", "resBlock2.bn2.bias", "resBlock2.bn2.running_mean", "resBlock2.bn2.running_var", "resBlock2.conv4.weight", "resBlock2.conv4.bias", "resBlock2.bn3.weight", "resBlock2.bn3.bias", "resBlock2.bn3.running_mean", "resBlock2.bn3.running_var", "resBlock2.conv5.weight", "resBlock2.conv5.bias", "resBlock2.bn4.weight", "resBlock2.bn4.bias", "resBlock2.bn4.running_mean", "resBlock2.bn4.running_var", "resBlock3.conv1.weight", "resBlock3.conv1.bias", "resBlock3.conv2.weight", "resBlock3.conv2.bias", "resBlock3.bn1.weight", "resBlock3.bn1.bias", "resBlock3.bn1.running_mean", "resBlock3.bn1.running_var", "resBlock3.conv3.weight", "resBlock3.conv3.bias", "resBlock3.bn2.weight", "resBlock3.bn2.bias", "resBlock3.bn2.running_mean", "resBlock3.bn2.running_var", "resBlock3.conv4.weight", "resBlock3.conv4.bias", "resBlock3.bn3.weight", "resBlock3.bn3.bias", "resBlock3.bn3.running_mean", "resBlock3.bn3.running_var", "resBlock3.conv5.weight", "resBlock3.conv5.bias", " [... cut ...] g_var", "module.upBlock4.bn3.num_batches_tracked", "module.upBlock4.conv4.weight", "module.upBlock4.conv4.bias", "module.upBlock4.bn4.weight", "module.upBlock4.bn4.bias", "module.upBlock4.bn4.running_mean", "module.upBlock4.bn4.running_var", "module.upBlock4.bn4.num_batches_tracked", "module.logits.weight", "module.logits.bias". 

Process finished with exit code 1
```

Did you save all required parameters during training? 
Or is there something wrong with my code?
Thanks for some feedback.

Sven

SvenMuc, 2021-07-13:
Short update. In the meanwhile I also tried the code from the trainer.py script.
I got the same errors.

```
model = SalsaNext(self.parser.get_n_classes())
...
torch.nn.Module.dump_patches = True
w_dict = torch.load(FLAGS.trained_model, map_location=lambda storage, loc: storage)

```

maskjp, 2021-07-13:
@SvenMuc ,

Did you try this [script](https://github.com/unmannedlab/RELLIS-3D/blob/main/benchmarks/SalsaNext/run_salsanext_eval.sh)?

SvenMuc, 2021-07-29:
I solved it. I had some conflicts with the installed cuda and pytorch libraries.

closed 2021-07-29 by SvenMuc