unmannedlab/RELLIS-3D #13: Mismatch between Ouster OS1 Point cloud in bag and bin files?

SvenMuc, 2021-07-29:
I'm trying to run the trained SalsaNext model directly on the provided rosbags.
But it seems, that the projections results and thus the model output is different between the point cloud stored in the bag and bin files (range of intensity, max distance, etc.). 

Into which coordinate system do I have to transform the point cloud to match the bin file format?
Did you limit the min/max range?
Did you applied any other data processing methods?

**Spherical Projection of bag file Point Cloud**
[image]

**Spherical Projection of bin file Point Cloud**
[image]

maskjp, 2021-07-29:
Hi, @SvenMuc,

When we convert the point cloud from rosbag to bin file, we only normalize the intensity value by dividing 255 to make the range is between 0 to 1. We didn't change the x,y,z values. 

When feeding the 2d data into the model, we [normalize](https://github.com/unmannedlab/RELLIS-3D/blob/7d92ce03407cdc630e3b14fe0d77422ec4199a0e/benchmarks/SalsaNext/train/tasks/semantic/dataset/rellis/parser.py#L229) the data by its [std and mean](https://github.com/unmannedlab/RELLIS-3D/blob/7d92ce03407cdc630e3b14fe0d77422ec4199a0e/benchmarks/SalsaNext/train/tasks/semantic/config/arch/salsanext_ouster.yml#L55). Did you do that?

SvenMuc, 2021-07-30:
Yes, the normalization I do. The whole pipeline is running perfectly with the bin files.
I added the division by 255 for the intensity as well. No the point cloud inputs are looking well.

But I still get bad results. It seems, that you did some coordinate transformations. The people walking behind the robot are positioned centrally in the projected images in the bin files. In the rosbags they appear at the outer left/right edge of the projection image.

Could you give me some further hints?

maskjp, 2021-07-30:
Hi, @SvenMuc,

Sorry, but can you try to divide the intensity by 65535. I remember that the range of the ouster intensity is larger than 255. 
Thanks!

phil0stine, 2021-08-05:
@SvenMuc Could you see if the PLY formatted clouds align with either of the projections you have produced here? The `bin` files were derived from the `ply` files, which were extracted directly from the rosbags.

chijie1998, 2021-12-29:
Hi, I am running into the same issue as well when trying to reproduce the benchmark using Salsanext. 
Just wanted to ask, do you convert the pointcloud from rosbag to PLY and then to KITTY .bin or rosbag to KITTY .bin directly?
Also which rostopic node is used? /velodyne_points or /os1_cloud_node/points?

maskjp, 2022-01-06:
Hi, @chijie1998,

Sorry for the late reply! We convert the point cloud directly from the rosbag using /os1_cloud_node/points topic. For the intensity value, we divided it by 65535.

I hope it helps!

chijie1998, 2022-01-11:
Hi, @maskjp 
Thanks for your reply, could you please provide the code/rosnode for converting and saving the /os1_cloud_node/points rostopic to Kitty .bin files? I tried my own converting method but I am getting some weird inferring results (i am using the 60s rosbag in 00000_00.bag). I am suspecting that it is the conversion from bag to bin which causes the problem because i am able to reproduce the benchmark without any problem using the .bin files provided.

Result by converting from rosbag :
[image]

Result using .bin downloaded directly :
[image]

Looking forward to your reply :)

maskjp, 2022-01-13:
Hi, @chijie1998 ,

I upload a [script](https://github.com/unmannedlab/RELLIS-3D/blob/main/utils/convert_ply2bin.py) to convert a ply file to a bin file. Please take a look at it.

For rosbag to bin file, it's the same. The key is to divide the intensity value by 65535.

I hope this helps!

If you still have problems, please feel free to contact me.

Best wishes!

chijie1998, 2022-01-13:
Hi @maskjp,

Thank you! Great news, I am able to get the results by convert the ros topic to pcd (Using ros_pcl) and then to .bin with intensity divide by 65535.

Just curious, how do you get the .ply file because I could'nt find any library in ros to this type of conversion. 
Also, can we do segmentation on the point cloud from velodyne using the same model? Or this model is specifically for Ouster point cloud and need to retrain for velodyne point?