unmannedlab/RELLIS-3D #15: Weird inference results

ljwpk, 2021-10-05:
Hi, I'm trying to run the pre-trained SalsaNext model with point clouds.

I created the environment with an anaconda.
All datasets for point clouds and lidar poses were downloaded and unzip.

Could you check my setup?

And then, I followed the readme file.

1. conda env create -f salsanext_cuda10.yml --name salsanext
2. conda activate salsanext
3. export CUDA_VISIBLE_DEVICES="1"
4. cd /path/to/benchmarks/SalsaNext/train/tasks/semantic
5. python infer2.py -d ~/Rellis -l ./prediction -s test -m ~/Rellis_pretrained

But outputs are weird. I attached the capture for the correlation matrix.

[image]

Thanks!

maskjp, 2021-10-05:
Hi, @ljwpk ,

Thank you for your interest in RELLIS3D. Your setup looks good. Could you give me more information? For example, the output of cell 35  in the Jupyter notebook, or share your prediction with me.

Thanks!

ljwpk, 2021-10-05:
Thank you for your reply.
This is the Jupyter notebook file and my prediction file.

Jupyter notebook: https://www.dropbox.com/s/y1tv5zgmwac4aeu/Untitled.ipynb?dl=0
Prediction result: https://www.dropbox.com/s/5l648jet38au0bd/prediction.zip?dl=0

ljwpk, 2021-10-06:
I downloaded it a few days ago.

I will try to create a new environment you gave.

LiDAR with Color Annotation PLY Format (Download 26GB) - Rellis_3D_os1_cloud_node_color_ply.zip
LiDAR SemanticKITTI Format (Download 14GB) - Rellis_3D_os1_cloud_node_kitti_bin.zip
LiDAR Annotation SemanticKITTI Format (Download 174MB) - Rellis_3D_os1_cloud_node_semantickitti_label_id_20210614.zip
LiDAR Scan Poses files (Download 174MB) - Rellis_3D_lidar_poses_20210614.zip
LiDAR Split File (75KB) - Rellis_3D_lidar_split.zip

Camera to LiDAR (Download 3KB) - Rellis_3D_cam2lidar_20210224.zip
Calibration Raw Data (Download 774MB)

Thanks!

ljwpk, 2021-10-11:
Not yet. I could not solve the problem.
Any guesses on this issue?

ljwpk, 2021-11-19:
@maskjp I solved this issue.
There was a related library version issue.

I tried to test different OS(Windows, Ubuntu).

1. Windows
My laptop consists of GTX1650 with CUDA 10.1(reqirements.txt crashed for PyTorch).

1) I remove PyTorch and torchvision in reqirements.txt
2) Install requirements. pip install -r reqirements.txt
3) Install PyTorch. conda install pytorch==1.6.0 torchvision==0.7.0 cudatoolkit=10.1 -c pytorch

2. Ubuntu
Machine learning computers had two RTX 3090 with CUDA 11.2.
When I tried to install it, I got the same issue.
1) & 2) are same.
3) conda install pytorch torchvision torchaudio cudatoolkit=11.3 -c pytorch

Thank you, maskjp.

maskjp, 2021-11-22:
Great! I am glad you solved your problem and thank you for providing this important information to us!

Thanks!

Best wishes!