unmannedlab/RELLIS-3D #20: LiDAR Annotation Semantic ROS Bag

DLonng, 2021-12-21:
Hi, Thanks for providing this dataset.

I want to use 
  - /os1_cloud_node/points
  - pylon_camera_node_label_color
  - /tf 
to build a semantic gridmap as a ground truth, but I don't know how to get such a bag.

Or, how to convert LiDAR Annotation SemanticKITTI Format to a ros bag file?
- /semantic_label_color_cloud
- /tf
- odom, gps, imu ...

Thank for your replay. :)

maskjp, 2021-12-21:
Hi, @DLonng,

Thank you so much for being so interested in our dataset. 

Every LiDAR point cloud in the rosbag has a corresponding semantic label and [pose](https://drive.google.com/file/d/1V3PT_NJhA41N7TBLp5AbW31d0ztQDQOX/view).

You can read the semantic label and put the information in the ROS message like the intensity information in the point cloud message. You can take [ouster lidar code](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/include/ouster_ros/point.h) as an example.

And use the pose file to register the point cloud. Or you can use other lidar-based slam algorithms to construct the map.

I hope this helps!

Best wishes!

DLonng, 2021-12-21:
ok, thanks :)

DLonng, 2021-12-21:
ok, thanks :)