unmannedlab/RELLIS-3D #23: Semantic segmentation labels on stereo frames

yqsony, 2022-01-20:
I am interested in semantic segmentation using stereo cameras. If I understand correctly, the semantic annotation labels are on the monocular RGB camera (pylon), but not the left or right camera in the stereo set. Is there any way that I can project the semantic annotation labels on to the left camera of the stereo set?

maskjp, 2022-01-25:
Hi, @yqsony,

Sorry for the late reply! We currently don't have the transformation between stereo and Balser cameras, so you can't project it on the camera. But we are working on it. We will provide it when we complete it. Appreciate your patience!

Best wishes!

maskjp, 2022-07-02:
Hi, @yqsony , @Alt216 
Sorry for the late reply. We recover the transformation using SFM. It's not perfect but might be usable.
Here is the [link](https://drive.google.com/file/d/1cP5-l_nYt3kZ4hZhEAHEdpt2fzToar0R/view)
I hope it helps!