unmannedlab/RELLIS-3D #25: Basler camera exact location

norbertmarko, 2022-03-09:
Hi!

I would like to ask if you could provide the exact location of the Basler camera. I am doing research with IMU and Camera data, and the exact mounting position would be crucial to use this dataset for the task. I read the paper on the dataset but the exact location is not indicated. If you also know the camera center and the mounting angle of the camera that would help me tremendously to further my research using this dataset.

Thanks in advance!

maskjp, 2022-03-16:
Hi, @norbertmarko,

Sorry for the late reply!

We have a description file [here](https://github.com/unmannedlab/cartographer/blob/main/src/platform_description/urdf/warthog.urdf.xacro). But it includes imu position between Velodyne and ouster. By combining the [camera Ouster relative position](https://drive.google.com/file/d/19EOqWS9fDUFp4nsBrMCa69xs9LgIlS2e/view), you can get the location.

I hope this helps!

Best wishes!

norbertmarko, 2022-03-25:
Hi @maskjp!

Thank you for your reply! I have few more questions about this since I am not familiar with the notations.

When I open the description file, there are positions (`xyz`, `rpy`) for Velodyne and Ouster as you suggested and their parent link is the `imu_link`. I'm guessing that is the relationship that you mentioned? If I follow the parent links back (imu_link --> top_chassis_link --> chassis_link)  I arrive at the values `<origin xyz="0 0 0" rpy="0 0 0"/>`. 

My question are: 
1. What position does the origin refer to on the paper's illustration on your mounted configuration?
3. Are the `xyz` values given in centimeters like in your paper?
4. Does the 'Pylon' camera in the config.yaml file refer to the Basler camera?
5. How are those values in the yaml files relate to the `.xarco` file that you linked and how can I calculate the exact Basler camera center position from that?

Sorry if there are too many questions, but these are not entirely clear and I'm guessing that this information could be valuable for a lot of people in the future.

Thank you very much!

maskjp, 2022-04-03:
Hi, @norbertmarko,
Sorry for the late reply. I have been occupied by a deadline recently. 
1. The "chassis_link"'s XYZ is "0,0,0".
2. Yes, the unit is centimeters.
3. Yes, Pylon camera. 
4. You can use the calibration between camera -> Lidar -> Imu to get the camera's relative position to imu. 
I hope this helps!

norbertmarko, 2022-04-25:
> Hi, @norbertmarko, Sorry for the late reply. I have been occupied by a deadline recently.
>
>     1. The "chassis_link"'s XYZ is "0,0,0".
>
>     2. Yes, the unit is centimeters.
>
>     3. Yes, Pylon [...]

Hi @maskjp ,
I calculated the transform from chassis link to Pylon camera, that is good so far. However there seems to be no way to calculate the exact physical location. There is a figure (1) in your paper and on your website about your platform configuration, but there are missing values, I included it for my questions:

1. What **physical point** the chassis_link refers to? **Is it one corner of your item profile? If yes, which one?** I know it's 0,0,0 but  what physical point in the real world are you using? _(I circled one candidate with red-yellow on the image below for an example of what am I asking.)_

2. The **height of the whole configuration** (without the robot)? I marked what I mean with the **red arrow on the image below**.

3. The height from the ground to your config. If you know this value (marked with black arrow on the picture below) that would help a lot, but I suppose that could be calculated from the 3D model on the Clearpath website.

Picture for my questions: https://imgur.com/a/WW5zgi2

If I combine these values with the already calculated transform, I could calculate the exact camera location in the physical world. If we sort this out I will leave the exact steps here for future use.

Thanks again for your help!

maskjp, 2022-04-29:
Hi, @norbertmarko ,
Sorry, it took me some time to access the robot, it's not around me. The following is the answer to your questions. 

1, I am not sure what physical point the chassis_link refers to? The robot urdf comes from clearpath. According to the urdf file, the chassis_link corresponding to [chassis.stl mesh](https://github.com/unmannedlab/cartographer/blob/main/src/platform_description/meshes/chassis.stl). The physical point can correspond to the center of the mesh.
```
  <link name="chassis_link">
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <mass value="100"/>
      <inertia ixx="4.83333333333" ixy="0" ixz="0" iyy="15.9375" iyz="0" izz="15.9375"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://platform_description/meshes/chassis.stl"/>
      </geometry>
      <material name="black">
        <color rgba="0.15 0.15 0.15 1.0"/>
      </material>
    </visual>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://platform_description/meshes/e-stop.stl"/>
      </geometry>
      <material name="red">
        <color rgba="0.9 0.0 0.0 1.0"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://platform_description/meshes/chassis-collision.stl"/>
      </geometry>
    </collision>
```

2, The first floor is 42 cm height and the second floor is 7 cm height.
3, The height from the robot is 83cm. (Please refer to [here](https://clearpathrobotics.com/warthog-unmanned-ground-vehicle-robot/))

Hope these information can help!

Best wishes!