unmannedlab/RELLIS-3D #28: OS1 Lidar max range setting

78csai, 2022-05-01:
Can you pls provide the used setting for the maximum range of the OS1.
According to the datasheet more settings are possible or does it mean the accuracy decreases with increasing range?
[image]

maskjp, 2022-05-02:
Hi, @78csai,

I think the default maximum range of OS1 is 120m. We used the default setting. The performance depends on both range and Lambertian reflectivity. If 80% Lambertian reflectivity, the accuracy is high within 100 m. If you want high accuracy, you can only use data within 100 m. 

Hope this helps! 

Best wishes!

78csai, 2022-05-05:
thanks a lot!

closed 2022-05-05 by 78csai