unmannedlab/RELLIS-3D #30: Question about Lidar timestamp

FrankXue95, 2022-10-09:
Hi, a nice off-road dataset. I have three questions about Ouster point cloud.
(a) For Ouster LiDAR SemanticKITTI Format, could you please provide timestamp info of each lidar scan (it can not be always 100ms);
(b) When I visualize the labeled points, there are always some noisy annotations located in the grass (shown as the figures below), is this normal?
[image]
[image]
(c) What are the criteria that distinguish trees from bushes?  Sometimes the top half of a tree is labeled as a tree and the bottom half as a trunk, which confuses the algorithm.

[image]

I look forward to hearing from you soon.

With kind regards,
Frank Xue

FrankXue95, 2022-10-10:
By the way, I have another question. could you please provide the distance between the Ouster Lidar and the ground plane？Thanks

maskjp, 2022-11-14:
Hi, @FrankXue95,

Sorry for the late reply. You can get the timestamp from the rosbag. Each scan corresponds to a ROS message.

We don't have the distance between the ouster lidar and the ground plane.

I hope it helps!

Best wishes!