unmannedlab/RELLIS-3D #36: Intensity Normalization

seungtae-Han, 2023-10-17:
Hello RELLIS-3D,

I am currently training a deep learning network for point cloud segmentation using Rellis-3d, and I am conducting inference on point clouds that I have obtained using my lidar sensor.
The lidar I use (both Rellis-3d and the lidar I use) is os1 64.

The lidar point cloud I have acquired has intensity values greater than 1.
However, the intensity in Rellis-3d is observed to be between 0 and less than 1, and I'm curious about how the values were normalized!
(For example, did you acquire lidar data and then complete Rellis-3d by dividing all intensity values by the largest intensity value?)

This is a visualization of the intensity distribution from a randomly selected bin file in the Rellis-3d dataset:
[image]

And this is my lidar data:
[image]

kasiv008, 2024-01-22:
The rellis 3D intensity data is normalized from 16 bit. You would need to multiply the data with 65535 to get the raw intensity value.

seungtae-Han, 2024-02-20:
> The rellis 3D intensity data is normalized from 16 bit. You would need to multiply the data with 65535 to get the raw intensity value.

Thank you!!! You've been a great help to me.