unmannedlab/RELLIS-3D #37: label distribution

lzbushicai, 2024-01-22:
[image]
Hello, I would like to calculate the label distribution of RELLIS3D. Can I obtain the label distribution data in the graph?

Thistoo, 2024-03-19:
I think you can obtain the ratio for total number of points in yaml file.
It's named 'content' I think.

lzbushicai, 2024-03-19:
Hi，I find an another problem in your datasets,
[image]
file:///tmp/MicrosoftEdgeDownloads/b88f61c0-9ed5-4f17-bd63-8f72f201a5fc/image.png
The first picture is a point cloud image of sequence03_165 frame. After aggregating 30 frame point clouds, I found that the orange obstacles shown in the picture have shifted after being superimposed. This may be because pose.txt is inaccurate.

At the same time, I projected the point cloud of this frame onto the image. The point cloud and the image do not seem to correspond well. See the figure below for details.
Maybe there is something wrong with the external parameters？？
[image]

phoenixGG, 2025-11-13:
Hi, the RELLIS-3D dataset is a great piece of work. I also want to obtain detailed data metrics distribution. Where can I get it? Thanks. @maskjp @phil0stine @maskjptamu @lzbushicai